jerabaul29/OpenMetBuoy-v2021a #4: use RTC instead of 1s interrupt + volatile memory

jerabaul29, 2021-12-10:
For now I am setting the RTC to generate interrupts at 1s intervals, and counting with an IRC the number of seconds. As pointed by @gauteh , I could probably just do everything directly with the RTC, without using an additional IRC. That may require some more HAL calls / setup.

gauteh, 2021-12-10:
* https://github.com/sparkfun/AmbiqSuiteSDK/blob/e280cbde3e366509da6768ab95471782a05d2371/boards/apollo3_evb/examples/rtc_print/src/rtc_print.c#L278
* https://github.com/sparkfun/AmbiqSuiteSDK/blob/e280cbde3e366509da6768ab95471782a05d2371/boards/apollo3_evb/examples/rtc_print/src/rtc_print.c#L326

You don't need all the TIMER stuff from that example.

jerabaul29, 2021-12-10:
Thanks :) . Agree that looks better, but also a bit worried that there are many moving parts, a bit afraid that I can easily shoot myself in the foot if I set a non self consistent bunch of options at the same time.