jerabaul29/OpenMetBuoy-v2021a #22: wave sensor (9dof): where to position it and how to fix breadboard?

CatherineTaelman, 2022-03-09:
Should the wave sensor (9dof) be placed right next to the antenna or rather on the opposite side of the box? 
And how is the breadboard attached on the wall of the box? (glue? duct tape?) It is not visible on the pictures in the manual. Thank you!

jerabaul29, 2022-03-09:
Location and orientation of the 9dof should not play a role. I have it on the same side of the batteries as the antenna, but a bit apart from each other. Anyways, GPS/Iridium is never active at the same time as the 9dof as things are now, so it should not play a role. I personally wrap it in duct tape (no naked metal), tape it to the wall, and add a dot of glue after testing :) .

Does that sound reasonable? :)

CatherineTaelman, 2022-03-09:
Sounds good, we will also have the antenna and wave sensor on the same side. There is enough space for that in the big boxes

jerabaul29, 2022-03-10:
Updated the instructions.

closed 2022-03-10 by jerabaul29