jerabaul29/OpenMetBuoy-v2021a #25: STEVAL-MKI207V1 instructions for connections

CatherineTaelman, 2022-03-09:
A question about the intructions found here: https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/instrument_hardware/ST_eval_instructions.md . 

**Is it correct that we should be connecting the red qwiic cable (3.3V) to pin 4 for SCX and pin 5 for SDX?** 

Reason for confusion is the following:
From the diagram found online (https://www.st.com/en/evaluation-tools/steval-mki207v1.html), it seems like SCX is connected to two pins (P4 and P24), but the connection to P24 had a resistor that says DNM, which we assume means 'do not mount'. So we concluded that the red cable should go to P4 for SCX. Same holds for SDX (with P5 and P17 as connected pins), where P17 had a DNM resistor, so the red cable for SDX should go to P5. 

Might be nice for future users to add the schematic diagram of STEVAL-MKI207V1 as a link on 'ST_eval_instructions.md' page :)

jerabaul29, 2022-03-09:
Great to see you go in such details to check :) . Many thanks for raising this issue here :) .

TLDR: yes, I think you are right, and thanks for pointing this out :) .

cc:ing @gauteh here :) (Gaute should get credits for finding that the chip was available in this way, and helping me find out how to correctly wire it :) ).

A side note: @gauteh had looked in details at the datasheet for the ISM330DHCX itself for knowing which pins to connect or not, see the datasheet available at: https://www.st.com/en/mems-and-sensors/ism330dhcx.html section 3.1:)

Regarding your question: I *think* (I do not have the chip under the hand to double check; if you and / or @gauteh can confirm it would be perfect :) ) that you are right and that the pins to use are the pins connected to R3 and R5, ie pins P4 and P5, ie the SCx and SDx on the side of VDD and VDDIO rather than on the other side. I agree that it looks like the pins SCx and SDx on the other side of the breakout, i.e. pins P17 and P24, are only "connected" through a non-mounted resistor, ie not connected and not active. This is indeed very confusing, and I will have to update the instructions, to make it clear that the pins on the side of VDD and VDDIO should be used, and NOT the pins on the other side, when connecting SCx and SDx.

I was actually not really aware of this, went a bit fast and was a bit sloppy myself, and I just connected to some "randomly chosen" SCx and SDx on my breakouts... It turns out that, since I am "luckily lazy", I connected the SCx and SDx corresponding to pins P4 and P4 to +3.3V, since these are the 2 close to VDD and VDDIO, so I could cut a shorter jumper ^^ . This is actually (barely) visible on the picture https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/instrument_hardware/jpg/STEval_PCB.jpg . If you zoom there you can see that I did let the SCx and SDx on the SDA/SCL side unconnected, and I connected instead the SCx and SDx on the VDD and VDDIO side. Pure luck, but need some now and then ^^ :) .

As a side note:

- I tested relatively extensively my setup corresponding to the description above, and it seems to work very well
- I had initially not read the datasheet as I should have, and not connected SCx and SDx at all, and things were working fine if I remember well (though as @gauteh had pointed out then, floating inputs that should be clamped was not a good idea)

Having an extra pair of eyes on this would be great :) . So if you can read this being quite critical / not "trusting", and double check and confirm, would be great :) .

Let us confirm, keep the issue open, and I will update the readme :) .

jerabaul29, 2022-03-09:
@gauteh  I thought so, that it does not matter which SCx and which SDx was connected as long as one of each was connected to 3.3V, but looks like as @CatherineTaelman points out from the STeval datasheet and how the PCB looks like that the SCx and SDx on the side of the SCL / SDA pins are just not connected to the chip in the center at all - while the SCx and SDx on the side of VDD and VDDIO do are connected to the chip in the center.

jerabaul29, 2022-03-09:
Would need a breakout under my hand (all mines are mounted in the boxes that are now sealed and glued) and a multimeter to check, though :) .

CatherineTaelman, 2022-03-09:
Thanks for double-checking and confirming! We also noticed indeed that the SCx and SDx on the side of the SCL / SDA pins are not connected at all to the chip. So we are going for the connections to P4 and P5 :)

jerabaul29, 2022-03-09:
Perfect, then this is confirmed, and I will update the md instructions file and let you know :) .

gauteh, 2022-03-09:
Great, that is a very important bit of information. Good catch!

jerabaul29, 2022-03-09:
Many thanks again, updated the readme here: https://github.com/jerabaul29/OpenMetBuoy-v2021a/commit/c627122b078af8c59dee82092ee670f9348295b6 .

Btw @gauteh , is it ok if I add your schematic (see under) to the md file (with credits of course :) ):

[image]

gauteh, 2022-03-09:
> Many thanks again, updated the readme here: c627122

Of course 😊

CatherineTaelman, 2022-03-09:
> Many thanks again, updated the readme here: [c627122](https://github.com/jerabaul29/OpenMetBuoy-v2021a/commit/c627122b078af8c59dee82092ee670f9348295b6) .
> 
> Btw @gauteh , is it ok if I add your schematic (see under) to the md file (with credits of course :) ):
> 
> [image]

That schematic is really nice for the readme! :)

tnn77, 2022-04-24:
Dear @jerabaul29 
Sorry, I have really lagging on this, and now I need to move fast for the AWI prep. 

May I ask about the resistor R1? So have you guys had a bit more experience now with this chip and the resistor R1 is not necessary?