jerabaul29/OpenMetBuoy-v2021a #28: cheaper bundles from Rock7 when buying more?

jerabaul29, 2022-03-10:
@CatherineTaelman if you do get a cheaper bundle when buying more from Rock7 please let me know (how much do you need to buy / how much cheaper :) ).

jerabaul29, 2022-03-11:
FYI @jvoermans @tnn77 : may allow to reduce the costs a bit if buying for many instruments at once :) .

tnn77, 2022-03-15:
@jerabaul29 
which cost are we talking about here??? Iridium fee or the Rock7 Iridium module?

jerabaul29, 2022-03-16:
The cost of iridium credits bought from Rock 7. Ie the credits bundles. Seems it may be possible to buy even more at once to get a better price, by taking contact directly with Rock 7 sales department, but this is only a hypothesis.

tnn77, 2022-03-16:
Great. Thank you so much for the info. We will try that for next time.

jerabaul29, 2022-03-31:
Just to double check @[CatherineTaelman](https://github.com/CatherineTaelman) , did you try to negociate a cheaper bundle, or just bought it "as is" for this time? :)

CatherineTaelman, 2022-03-31:
I havent asked, we decided to go for the 20 000 credits bundle for now, as a start. But the company is very responsive to emails, so it is easy to get in contact with them for any requests:)

jerabaul29, 2022-03-31:
Ok. Would need to compute the cost :) . Is it easy to switch offering?

jvoermans, 2022-03-31:
I think it isn't much of a problem to switch, but the data plans are for 12 months, so doesn't work for me. I'm asking now for larger bundles...

jerabaul29, 2022-03-31:
Aaah, ok, yes 12 months binding is not so good. Bigger data plans would be better :) .

jerabaul29, 2022-04-05:
Many thanks @jvoermans for looking into this. So it looks like no real interest in it, no real cost cuts to get. Closing.

closed 2022-04-05 by jerabaul29

tnn77, 2022-05-26:
@jerabaul29 @jvoermans 
FYI - it was quite easy to set up NET30 terms between Ground Control and AWI. This allows you to buy line rental and credits without paying upfront, and every 30 days they will do a sweep, collate all spends on the account and send through an invoice.

Something to consider for large scale deployments.

I think all you need to provide them is the approximate number of line rentals and credits you'd expect to purchase.