jerabaul29/OpenMetBuoy-v2021a #31: request for specific binary files of firmware

CatherineTaelman, 2022-03-24:
Hi Jean,

We will be deploying 3 drifters on icebergs. These drifters have only GPS (no wave sensor). 
We would like to have two different sampling and transmission rates because of two different scenarios:

- One drifter will be deployed on a floating iceberg. For this one, we would like the GPS location to be sampled every 30 mins and transmission approx. every 3 hours.

- 2 drifters will be deployed on grounded icebergs. For them, it is crucial that we extend the battery lifetime as much as possible (we'd like to still have them working after 2 months). We were thinking of: sampling GPS location every 12 or 24 hours and transmitting every 48 hours. Does that sounds reasonable to you? 

The question is: could you compile 2 different binary files with these specific sampling and transmission rates? 

Thank you so much already!

jerabaul29, 2022-03-24:
That sounds reasonable :) .

The compiled binaries are available in the repo now.

I have never used GPS sampling rate less often than 1 hour before now. This should not cause any issue that I know of, but may be a good idea to test a bit thoroughly the drifters with 12hr resolution and 48hr latency :) .

jerabaul29, 2022-03-24:
PS: how are things going, have you been able to test that all works well? :) .

CatherineTaelman, 2022-03-25:
Hi and thank you so much for providing the new binary files! We are right now doing the final assembly of all the drifters. There are some struggles with getting the Iridium package through the university finance system, so that has delayed the testing fase. But I will be putting out 3 drifters this evening to start testing over the weekend. I will let you know on Monday if everything works well:)

CatherineTaelman, 2022-03-25:
small other question: what is the transmission rate in 'steval_gps_waves_drifter.bin'? That is the one we will use for the 17 drifters in large boxes.

CatherineTaelman, 2022-03-25:
From the logging file I read out in Arduino IDE,  I assume the transmission rate is every 2 hours (after initial GNSS fix fail, it goes to sleep for 7200 seconds). 

When checking the other variables in 'steval_gps_waves_drifter.bin', I noticed the variable 'interval_between_wave_spectra_measurements: 7200'. So that means it measures wave spectra every 2h, right? In the description of the binary files it says every 3h, just want to check what is the correct value :)

jerabaul29, 2022-03-25:
Sorry, away from my computer.

The 'true' answer is, as you say, the serial output :) . The 7200s after failed GPS fix is not the frequency of GPS measurements, but a way to save battery if not GPS reception, for example if switched on while in a container by mistake.

If I remember well, the GPS measurements frequency should be 30 minutes, and transmission every 2 hours. That should be visible in the serial output.

If you are in doubt, just copy paste here the start of the serial output, and I can go through it for you here :) .

CatherineTaelman, 2022-03-25:
Check, GPS every 30 mins and transmission every 2h is indeed what the serial output says :) Thanks for double-checking!

CatherineTaelman, 2022-03-25:
[image]

jerabaul29, 2022-03-25:
This will actually be around every 3 hours, since 6 messages minimum per GPS packet :) .

CatherineTaelman, 2022-03-25:
Okay I see, thank you!

jerabaul29, 2022-03-31:
Seems to be confirmed to work well, closing :) .

closed 2022-03-31 by jerabaul29