jerabaul29/OpenMetBuoy-v2021a #33: smart "freeze in" vertical mount

jerabaul29, 2022-03-31:
@[CatherineTaelman](https://github.com/CatherineTaelman) if you are ok to send a picture of your smart "freeze in" vertical mount that would be great :) . I think that the idea could be used by both @jvoermans and @tnn77 to avoid problems with being snowed in.

tnn77, 2022-04-05:
Hi, @jerabaul29 .
I'm very curious about this. If not a picture, is it possible to share a sketch of this mount?

jerabaul29, 2022-04-06:
Just putting a small hand drawing, from what I understood from @CatherineTaelman , all credits going to her :) (hope it is ok that I upload here a sketch of what you were suggesting? if not, let me know and I will remove it at once).

[image]

CatherineTaelman, 2022-04-06:
the long bolts going through the PVC pipe and connecting the two metal hooks to the pipe, the 2 small bolts are meant to attach the drifter box to the metal hooks

jerabaul29, 2022-04-06:
Many thanks for the extra pictures, looks really good :) .

When setting these up, I think that a tradeoff question is "how high above the ice" should the box be: the higher above the ice, the less risk for getting buried down by snow (which is really great for iridium communications), but the higher the risk to i) be spotted by bears, ii) start vibrate if there is a lot of wind / the tube is not very stable. Wonder where the optimal value of this tradeoff goes: 5cm above the ice? 10? 15? 20? even more (I think 20cm is a maximum though)?

tnn77, 2022-04-06:
Dear @CatherineTaelman @jerabaul29 
thank you so much. this is a nice idea. 

I have a similar question as well. I was wondering how much elevation we may gain with this approach. I guess the target height would depend a lot on the expected measurement duration and snow conditions.

jerabaul29, 2022-04-22:
Closing now, will add in a dedicated readme file, feel free to re open if necessary / if you want to discuss further :) .

closed 2022-04-22 by jerabaul29