jerabaul29/OpenMetBuoy-v2021a #35: A version without the Qwiic Power Switch

gauteh, 2022-04-04:
The [Qwiic Power Switch](https://www.sparkfun.com/products/16740) is out of stock with no dates for estimated availability. Would it be possible to have a version where it is not necessary, e.g. using compile flags?

Best Regards, Gaute

jerabaul29, 2022-04-04:
That may be possible, but with a large power consumption penalty. Have you tried to send a message to Sparkfun asking about availability? For now, I can provide you with a few from my "personal" stock :) .

gauteh, 2022-04-04:
The IMU uses about 1.5mA, so that should translate to at most twice the present consumption - probably much less.

jerabaul29, 2022-04-04:
Just got confirmation from SF (I tagged you there :) ) that it is in the production queue :) . In the meantime you can get a few from me, and give me some back when you get to order from SF :) . These are so cheap that it may be worth ordering a 10 or 20 at once to have a bit in stock :) .

gauteh, 2022-04-04:
Great! If I can borrow some from you I am happy!

jerabaul29, 2022-04-05:
The conclusion is that, for now, I will not develop one: closing.

closed 2022-04-05 by jerabaul29