jerabaul29/OpenMetBuoy-v2021a #36: frequency vs period dyslexia

jerabaul29, 2022-04-05:
It looks like I have suffered a case of frequency vs period dyslexia :) .

There (and in all files where this is repeated):

https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/0b4909a25e8a84f1c9a2b364c7d89899f034a9b2/legacy_firmware/firmware/standard_gps_waves_drifter/wave_analyzer.cpp#L233-L234

What I am *really* computing is ```fz``` and ```fc```, NOT ```Tz``` and ```Tc```. So, there is a factor ```1.0 /```. This is a general error, it would be error prone to correct it everywhere and to recompile the (otherwise fine) firmware, so I think that the best method is to "mark it as a quirk", and fix this in the decoder instead, by inverting the values to recover the correct value.

tnn77, 2022-04-05:
@jerabaul29 
ah sorry, I use the decoder to obtain spectrum only, so I did not notice this.

jerabaul29, 2022-04-05:
No worries :) . Yes indeed, all of this is anyways only derived from the spectrum that we are transmitting in full, so this is just redundant - it is at least as good to work directly with the spectrum :) .

CatherineTaelman, 2022-04-05:
Good to know, thank you Jean!