jerabaul29/OpenMetBuoy-v2021a #37: previous version of the binary steval_gps_waves_drifter.bin was not always switching off the imu

jerabaul29, 2022-04-07:
@[CatherineTaelman](https://github.com/CatherineTaelman)  @jvoermans and @tnn77  , the previous version of https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/legacy_firmware/binaries/steval_gps_waves_drifter.bin was not always switching off the IMU after use. This did not result in any issue with the functioning of the instrument itself, but will increase power consumption. The firmware version I just pushed fixes this issue, and should both work well (as the previous one), and reduce power consumption by making sure the IMU gets switched off.

- I will fully test now and let you know / confirm that it works well
- I recommend that you update the firmware of the instruments that are not sealed and were using the old steval_gps_waves_drifter.bin firmware , if possible. If not possible to update, this is not a really critical issue, as the power consumption increase was not too bad, but still, better to be power effective and get longer battery life :) .

jerabaul29, 2022-04-07:
Confirm the updated firmware works perfectly well :) .

jerabaul29, 2022-04-22:
Closing for now, feel free to discuss further if anything is unclear :) .

jerabaul29, 2022-04-22:
@CatherineTaelman not sure if you received the notification, just doubling on it just in case :) .

closed 2022-04-22 by jerabaul29

CatherineTaelman, 2022-04-22:
Hi Jean, thanks for noticing me about this issue, it has slipped through my attention with all the other Github discussions recently.. I will download the updated binary file and upload it on the drifters here on the ship:)

jerabaul29, 2022-04-22:
Sounds good :) . My advice: download it and test it on one instrument first (you never know ^^ ), anyways you have good time until you reach the ice I guess? :)

CatherineTaelman, 2022-04-22:
okay, I will try it out on one instrument tonight (if time allows), we are leaving harbour any moment now and will reach the ice in 1-2 days approx. So should leave a little bit of time to test the functionality with the new code :)