jerabaul29/OpenMetBuoy-v2021a #53: parser script not working on Windows

jerabaul29, 2022-05-19:
From user feedback:

"
AttributeError: module 'time' has no attribute 'tzset'.
"

This is a windows vs. linux compatibility issue:

https://stackoverflow.com/questions/1988182/why-my-time-class-has-not-tzset-attribute

```
See [the docs](http://docs.python.org/library/time.html?highlight=tzset#time.tzset) for tzset: they clearly say

    Availability: Unix.

so you would have it in, say, MacOSX, Solaris, or Linux, but not on Windows.
```

To run the decoder, use either a linux system, or run on windows through the WSL - windows subsystem for linux.

I do not have access to a Windows machine, and I will only provide support for linux myself (specifically, Ubuntu distros, 20.04 or newer).

jerabaul29, 2022-05-19:
For WSL, see for example https://pbpython.com/wsl-python.html and similar. Using Ubuntu OS in WSL you should be able to reproduce the exact same kind of environment I use, and things should work.

jerabaul29, 2022-06-01:
Also possible to use an Ubuntu Virtualbox.

jvoermans, 2022-09-13:
Bit late, but you can just remove 'tzset' in windows, then run script through Spyder or equivalent, works for me...

jerabaul29, 2022-09-13:
Thanks for the update @jvoermans . Be careful / be aware though that, if you do not use the tzset, you may have bad surprises about which time zone python may choose for you / be using under the hood... That may result easily in some "corruption" of timestamps by shifting the time value by a few hours when converting stuff back and forth later in your scripts. I think a better option would be that you find how to set up the time zone to utc on windows and continue using tzset, but with a windows compatible backend :) .

jvoermans, 2022-09-13:
Noted :)