jerabaul29/OpenMetBuoy-v2021a #60: Binary for measuring waves every hour

bjorkqvi, 2022-07-05:
Is there a possibility to get a precompiled binary that measures and sends wave data every hour and GPS every 30 minutes?

Thanks!

bjorkqvi, 2022-07-05:
Thanks! I was thinking that the GPS could send every 30 minutes. If we loose the buoy and need to find it, then 1 hour is a long time. We are only measuring for a week, so battery consumption is not a concern. But this is not a big issue. 

I will have time to test this program tomorrow on the roof.

jerabaul29, 2022-07-06:
I can generate one more version tonight sending GPS every 30 minutes. The issue then is that if the reception is a bit bad, there are so many iridium messages that need to be transmitted that the instrument may miss some wave spectra by getting 'behind schedule'.

bjorkqvi, 2022-07-06:
Ok, the reception might definitely be bad where we are going. Thanks for this heads up, then I will use the one that only sends every hours, since we don't want to risk loosing any wave data.

closed 2022-08-05 by jerabaul29