jerabaul29/OpenMetBuoy-v2021a #64: antenna cable, impedance, and L-shaped connector

jerabaul29, 2022-08-31:
Can the SMA cable cause loss of signal due to unmatched ampedance? The feelings from Tak seems to say that yes.

jerabaul29, 2022-11-22:
ie this should be quite a bit better than SMA cable: https://no.rs-online.com/web/p/coaxial-adapters/7009600?cm_mmc=NO-PLA-DS3A-_-google-_-CSS_NO_NO_Connectors_Whoop-_-(NO:Whoop!)+Coaxial+Adapters+(2)-_-7009600&matchtype=&pla-339426036708&gclid=Cj0KCQiA4OybBhCzARIsAIcfn9llkMNhJM6O4Qb0oPlEeYkI1M0Klh2eP4GqLsue_bemLERTMFB3dzMaAlfHEALw_wcB&gclsrc=aw.ds .

jerabaul29, 2022-11-22:
L-shaped 90 degrees, SMA male to SMA female, typically 50 Ohm impedance.

jerabaul29, 2022-11-22:
this can be used to hold the antenna, either directly looking up if possible, or else with a bit of an angle

jerabaul29, 2022-11-22:
That looks like a less expensive alternative: https://no.farnell.com/amphenol/sma5072a2-3gt50g-50/adaptor-sma-plug-sma-jack-50ohm/dp/1111341

jerabaul29, 2022-11-22:
@gauteh may be relevant to use some too :) .

gauteh, 2022-11-22:
Thanks. In general it seems to work pretty well as it is also.

jerabaul29, 2022-11-22:
Yes, but according to Tak et co it was making a big difference (erratic GPS fix vs perfect GPS fix, no iridium vs reliable iridium) when they were testing in Tokyo with bad sky visibility; that may apply to storm conditions / large waves with lots of sea spray in the air / icing on the box as well - if there are a few dBs of RF gain to take for free, why not? :) ^^