jerabaul29/OpenMetBuoy-v2021a #69: alternative magnetic switch

jerabaul29, 2022-11-07:
Following the recommendation from @jvoermans , added the alternative for a more robust magnetic switch in the possible list of components, see:

- 1 x MK15-B-2-OE magnetic switch, normall close: 7 USD: https://www.digikey.com.au/en/products/detail/standex-meder-electronics/MK15-B-2-OE/2765323 , see datasheet at https://standexelectronics.com/viewer/pdfjs/web/viewer.php?file=https%3A%2F%2Fstandexelectronics.com%2Fwp-content%2Fuploads%2F2015%2F03%2FMK15_V02.pdf

@tnn77 , that may be a good alternative for you, in particular for planning for it if you want to have your own PCB :) .

jerabaul29, 2022-11-07:
Closing, as this was just to notify you, but of course if there is anything you want to discuss feel free to re open and discuss here :) .

closed 2022-11-07 by jerabaul29

tnn77, 2022-11-11:
@jerabaul29 
this is very nice. indeed, robust switch would be useful (we usually break a few during production). thanks!