jerabaul29/OpenMetBuoy-v2021a #76: write guide for people using the OMB from Labmaker

jerabaul29, 2023-03-06:
- prepare Rock7
- prepare decoder script
- update the firmware if you want
- put batteries in
- seal "real good"
- wait for a few messages, and run the parsing scripts
- if you want, ask us to host your data on our server

jerabaul29, 2023-03-07:
@AndreaGierisch let us list points I should have in a "Labmaker buoy user helpguide" :)

jerabaul29, 2023-03-07:
- current from only 1 source
- static electricity protection
- what firmware to use
- logging debug info over USB

AndreaGierisch, 2023-03-07:
> @AndreaGierisch let us list points I should have in a "Labmaker buoy user helpguide" :)

Sure :-) Here is a short brainstorming of points which I think could be nice to have in a Quick Start Guide or so:

- What is what when I look at the electronics?
- Create an account at Rock7 to a) buy subscription/credits and b) see the received data
- How to download and start Artemis-Uploader
- How to check whether there is (the correct) firmware installed?
- What the firmware can do, and in which cases I need to write my own firmware
- What batteries to buy?
- What magnet to buy - And can I use the OMB also without magnet (I guess so.)
- Do I have the 6-dof or 9-dof sensor (relevant for firmware, I think?)
- How to change frequency of GPS recordings / message transmission?
- Table with expected battery lifetime
- Table with credits per message (and recordings per message) and price of credits
- Dos and Don'ts like:
-  - never two power sources at the same time
-  - slow cool-down from room temperature to outside
-  - use desiccant bags (and recommendations which/how many?) 
-  - seal the box with silicon?
-  - How to best hold the electronics plate when taking it out from the box to reach the batteries
 
I think I found answers to many of these items already, but it would have been easier of course if all was collected in one place ;-)

jerabaul29, 2023-03-08:
- need to be in the dialout group to access usb port

akorosov, 2023-06-21:
Regarding the batteries: Does the the LabMaker version have parallel or serial connection in the baterry holder? We bough OMB from LabMaker and Lithium LH20 baterries with 3.6V. Can we safely insert them?

jerabaul29, 2023-06-21:
@akorosov the LabMaker version have parallel connected battery holders and expects 2 LSH20 in parallel yes :) . You can safely insert them, just be sure to respect polarity of course (should match the markings on both the PCB and the battery holders).

jerabaul29, 2023-06-21:
- always antenna on when RF modules in use

jerabaul29, 2023-06-22:
Let me know how the tests go / if there is anything I can do to help @akorosov :) . Will be happy to have a chat with you and / or some of your colleagues if I can help :) .

jerabaul29, 2023-06-22:
I tried to summarize at https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/main/end_user_manual/LabMaker_OMB_instructions.md ; if you want more information, either ask for specific additions here, or open pull requests :) .

closed 2023-08-08 by jerabaul29