jerabaul29/OpenMetBuoy-v2021a #80: decoder: sometimes wave message has datetime_fix, sometimes datetime_packet

gauteh, 2023-03-17:
Up until now we have only used datetime_fix, but it seems that there is sometimes datetime_packet. What is the difference?

https://github.com/gauteh/sfy/blob/main/sfy-processing/sfy/cli/collection.py#L113

gauteh, 2023-03-17:
https://github.com/gauteh/sfy/pull/119

jerabaul29, 2023-03-17:
These concepts were maybe not so well defined in my head initially, so this is maybe not so consistent over time / files, and I may unfortunately have been inconsistent, but I would say:

- a fix is actually getting a valid fix from the GPS
- a packet is a fix encoded into its binary string, that becomes a part of a message
- a message is one or several packets, with a tiny bit of metadata, encoded in binary string, that is sent over iridium

Maybe we should think about standardizing the terminology / concepts for future buoy developments? :)

gauteh, 2023-03-17:
So they are synonymous now and we can treat them the same? (I assumed one was the time of packet recording, and the other was the time of gps fix).

jerabaul29, 2023-03-17:
Sorry, you are right, we were discussing different aspects (ie hierarchy in the data vs individual data fields). Honestly, most of this was designed in 18hr sprints in 2021 while I was on the icebreaker, so there is no real "master plan and consistent engineering", sorry about that.

Yes, these are anyways always posix timestamps, associated with the data, either as part of a full GPS fix, as in

https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/c7c9c8c10c187cacb17ec91363d69f7dd16eff34/legacy_firmware/decoder/decoder.py#L127-L132

or as part as  a wave packet, which does not really contains a GPS fix per se, but includes the RTC UTC posix timestamp, as in

https://github.com/jerabaul29/OpenMetBuoy-v2021a/blob/c7c9c8c10c187cacb17ec91363d69f7dd16eff34/legacy_firmware/decoder/decoder.py#L140-L143

I think I was not very consistent when I coded all of this, sorry. If I should find time to recode all of this in a consistent way, I would use the name datetime_fix in the context of an actual GPS fix, while datetime_packet should be used in the context of getting the UTC posix time for for example a wave or thermistor packet, but not from a true GPS fix. At the present time, this is not really consistent, sorry.

jerabaul29, 2023-03-17:
My terminology is very badly chosen and has overlaps / inconsistencies. I think we should also standardize the terminology / concept regarding the name of posix timestamps coming from either a GPS fix or pulling the RTC.

jerabaul29, 2023-03-17:
(doing a ctrl-f on the file, I used datetime_packet only a couple of time when adding posix information from the UTC clock to wave and thermistor packets, though in the dataclass it is still called a datetime_fix ... not very consistent. :( . At the same time, all of this is brittle, I think I will, in the context of the OMB, keep it as is for now, and next time an OMB version gets developed, start from scratch being more consisten :) ).

jerabaul29, 2023-03-17:
Sounds good :) . Closing for now, feel free to re open if needed :) .

closed 2023-03-17 by jerabaul29