jerabaul29/OpenMetBuoy-v2021a #83: docker container to compile the arduino sketch

jerabaul29, 2023-05-11:


jerabaul29, 2023-05-12:
@gauteh see the commit here: https://github.com/jerabaul29/OpenMetBuoy-v2021a/commit/bd651fd55271dfa694805120ff1aab79d39cc817 . Let me know if you have any comments :) . We want to make it sure that it is easy for people

gauteh, 2023-05-12:
Looks good. It would be nice with automatic building of all firmwares, then you will have checks in place. An users could get custom firmware by just creating a PR 😁

jerabaul29, 2023-05-12:
Thanks :) . Hehe thanks for the tips, but I think this would be a bit too much automating given that all of this is a bit brittle / it is easy to set conflicting parameters in the firmware / I would prefer to only have .bin firmware versions that are either tested fully (best) or compiled by myself, ideally both. We can set up CICD firmware versions building with the OMB-v2025 ^^ :) .